MaslowCNC/Firmware #418: Error Message: Invalid Library found

jarid283, 2018-03-28:
I keep getting an error message of an invalid library.

Error Compiling for board for Arduino Mega etc.

Invalid Library its saying.

blurfl, 2018-03-28:
Can you give some more info? Which OS are you using? Which version of the Arduino IDE (found in the title bar of the sketch). Can you use the 'Copy error messages' button to capture the compile error, paste it into a text document, and attach it to a reply here (drag it into the reply window)?

BarbourSmith, 2018-03-28:
Are you still able to upload the firmware and if so does it work?

jarid283, 2018-03-29:
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\Jarid283\Desktop\cnc_ctrl_v1\cnc_ctrl_v1.ino:15:20: fatal error: Maslow.h: No such file or directory

 #include "Maslow.h"

                    ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
Invalid library found in C:\Users\Jarid283\Documents\Arduino\libraries\cnc_ctrl_v1: C:\Users\Jarid283\Documents\Arduino\libraries\cnc_ctrl_v1
Invalid library found in C:\Users\Jarid283\Documents\Arduino\libraries\cnc_ctrl_v1: C:\Users\Jarid283\Documents\Arduino\libraries\cnc_ctrl_v1

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

blurfl, 2018-03-29:
I think I see the cause. The sketch folder 'cnc_ctrl_v1' has more files than just the cnc_ctrl.ino file, but all those must stay together inside the cnc_ctrl_v1 sketch folder. The sketch folder doesn't need to be moved, though you can if you like, though 'Documents\Arduino\libraries\' might not be a good choice. Putting the sketch folder in 'Documents\Arduino\' will make it show up in the Arduino 'File/Sketchbook' menu, but when moved the sketch folder has to keep all the files in it.

jarid283, 2018-03-29:
Thanks...just got it!

BarbourSmith, 2018-03-29:
Wonderful!

Great diagnosis @blurl 😀 👍 👍

closed 2018-03-29 by BarbourSmith