MaslowCNC/Firmware #424: 3-axis movement (enhancement)

mrfugu, 2018-04-08:
re: https://github.com/MaslowCNC/Firmware/pull/423#issuecomment-379520149

a request to enable 3 axis movement mainly to begin looking for improvements via: 
The current cutting method plunges the router, unloading it prior to movement via the hole cut during the plunge.
 
Post plunge, the router and sled effectively start X/Y motion at 0 load, the Sled moves to full inertial loading by beginning X/Y motion, almost immediately, but not immediately, the Router experiences full bit loading, via contact at full depth of cut.

Additionally, the lone Z-axis movement makes a plunge cut using only the force of the sled perpendicular to the work surface, this is already low, and decreases to 0 as the angle of the machine approaches vertical, and can also change dramatically by sled weight  (amount and placement), sled surface area, z-assemblies, different routers, a whole host of variables make the force that the sled is held to the work surface one of the most important aspects to remove 'slop'  from what is the relatively accurate 'stable and constant' X/Y motion of a well calibrated machine. 

3 axis movement will bring: 

1) The ability to decouple router/bit load from the sled's inertia loading, which should be quantifiable as the difference in accuracy between 'pen vs router' testing of 2 sleds of the same weight/COG 

2) ability to create paths that predictably ramp loading on the router should increase sled stability and predictable response to changes in motion.  A more stable sled makes any predictions made of the machine's other moving parts more accurate. 

3) A more consistent and continual loading of the router will result in more optimal use of and behavior from all of the cutting aspects of the machine. Where the rubber hits the road as it were.  The Router, the End Mills, the Workpiece benefit the most from the router cutting in the manner it was designed to perform best, ie: as constant as is possible moderate loading, at speeds and feeds predicted/calculated based on the end mill optimal consumption of material. 

We're clearly 'very nearly there' =)

davidelang, 2018-04-09:
On Mon, 9 Apr 2018, mrfugu wrote:

> Apologies for misnaming simultaneous 3 axis movement. I mean X Y and Z in motion together. ramping into a cut. I have yet to see Z move ever without X-Y motion stopped completely.

That's a g-code/CAM setting then, people have done simultanious 3-axis movement.

The problem is that it's not coordinated, so if the feed rate isn't perfect, the 
cut doesn't come out right.

mrfugu, 2018-04-09:
ok, I'll investigate it further in Fusion, but via its simulator I think I had it correct last week.  I'll report any findings.

blurfl, 2018-04-09:
>The problem is that it's not coordinated, so if the feed rate isn't perfect, the cut doesn't come out right.

Good news! 3-axis G0 and G1 coordinated move works correctly with the current firmware. The speed is limited by the speed of the z-axis, but with G0 and G1 X, Y and Z track correctly, reaching 50% of the z depth at 50% of the XY travel and the full depth at the same instant that X and Y reach their destinations.

mrfugu, 2018-04-09:
yeah, looks like a fusion360 error on my part. closing, sorry for the false alarm, I just found that branch when looking to confirm or not that there was 3axis movement. maybe close down that branch?

closed 2018-04-09 by mrfugu