MaslowCNC/Firmware #426: This file does not cut correctly

BarbourSmith, 2018-04-09:
When cut this file does not cut correctly

[test cut square rounded corner.txt](https://github.com/MaslowCNC/Firmware/files/1891189/test.cut.square.rounded.corner.txt)

You can see that there is a bump over on the left hand side:

[image]

While it renders right in Ground Control:

[image]

Here is the log for a test cut of the file:

[log.txt](https://github.com/MaslowCNC/Firmware/files/1891194/log.txt)

And the forums conversation about the issue can be found here:

https://forums.maslowcnc.com/t/odd-cut-when-cutting-test-shape/3230/3

BarbourSmith, 2018-04-09:
That sounds like an overflow issue in the arduino to me, or a rounding issue when dealing with numbers that large

davidelang, 2018-04-09:
well, it's that and a bad design file (it should not be doing arcs instead of 
stright lines)

davidelang, 2018-04-09:
On Mon, 9 Apr 2018, Scott Smith wrote:

>> That sounds like an overflow issue in the arduino to me, or a rounding issue when dealing with numbers that large
>
> Looking at the gcode parameter values as they are handled in G2(), I see
> float extractGcodeValue(const String& readString, char target, const float& defaultReturn)
> returning more digits than were in the source gcode string. I added code to G2() to print the I and J values returned by extractGcodeValue() as rawI and rawJ:
>
> Sent: G3 X-15.1990 Y-10.880 I9563.8560000000 J0
> iCS executing G code: G3 X-15.1990 Y-10.880 I9563.8560000000 J0
> rawI 9563.855468750000
> rawJ 0.000000000000
>
> Haven't yet looked at why that might be, or whether it is related to this issue though.

That is floating point to fixed point precision loss. floating point doesn't 
keep that many bits of precision.

blurfl, 2018-04-09:
>That is floating point to fixed point precision loss

 float extractGcodeValue() should return a float, yes? Is it the case that the print statement doesn't correctly print it? Subsequent calculations seem to indicate that the extra digits are there.

BarbourSmith, 2018-04-09:
I think that the extra digits ARE there, the issue is that the Arduino cannot store the number 9563.856 using a floating point number so it stores the closest number that it can which is 9563.85546875. Because the numbers are being stored as binary (1001010...etc) and there are only so many 1 and 0, it's not possible to represent every number perfectly

davidelang, 2018-04-09:
On Mon, 9 Apr 2018, Scott Smith wrote:

>> Arduino cannot store the number
> I see what's going on. Is there a standard way to handle this gracefully?

you either have to accept the inaccuracy of floating points, or switch to fixed 
point math (which is sometimes faster, but much uglier to handle)

blurfl, 2018-04-09:
I can accept the inaccuracy, but I wonder if there's a programmatic way to flag that it is about to happen before the cut 'runs off the rails'?

davidelang, 2018-04-09:
keep in mind "running off the rails" is an error of 0.0006 in for the error in I

that amount of error will not cause the type of errors being described.

blurfl, 2018-04-09:
@davidelang , I'm not sure I follow?

davidelang, 2018-04-09:
[quote]
Sent: G3 X-15.1990 Y-10.880 I9563.8560000000 J0
 iCS executing G code: G3 X-15.1990 Y-10.880 I9563.8560000000 J0
 rawI 9563.855468750000
 rawJ 0.000000000000
[/quote]
the difference between the I that was used and the I that was specified is
9563.8560000000 -
9563.855468750000 =
0000.00053125"

a very small amount, and not likely to be enough to cause any noticeable errors (and not enough to cause the problem we are seeing)

it's more likely that trying to specify an arc that large is causing problems, not that the error in the floating point math is causing problems.

blurfl, 2018-04-09:
I see, yes the issue is probably in the trig calculations done with the numbers in Motion.cpp. Part of what I think you're saying is that it would be very hard to identify that situation by inspecting the gcode parameter values themselves, yes?

davidelang, 2018-04-09:
On Mon, 9 Apr 2018, Scott Smith wrote:

> I see, yes the issue is probably in the trig calculations done with the 
> numbers in Motion.cpp. Part of what I think you're saying is that it would be 
> very hard to identify that situation by inspecting the gcode parameter values 
> themselves, yes?

I'm saying that the difference in the I value that we are seeing due to floating 
point numbers isn't the real problem

BarbourSmith, 2018-04-10:
I think that very slight rounding error when run through the sin and cos functions which define the movement could easily result in the bump we are seeing when multiplied by a radius that large

blurfl, 2018-04-10:
>a radius that large

That's an interesting point. Would that be the item to watch and flag?

BarbourSmith, 2018-04-10:
Good question, I'm not sure! 

I think it's pretty easy to say that any gcode which uses a radius of anything over a few thousand inches is asking for trouble.

Very astute observation @blurfl . I vote that we flag large radius. We can refine the technique based on if it's catching files which are OK or if it's not catching files which are trouble.

Let's word the message in such a way that the user knows that they can still run the file and it will probably be mostly OK, but that there were some warning signs

davidelang, 2018-04-10:
Let's start of by making some sample g-code files with large radius curves and 
see if they have problems. Make sure it's not something else odd (and we can 
check if it's a floating point resolution issue at the same time)

blurfl, 2018-04-10:
Looking deeper, the radius doesn't seem to be the problem. Instead, there is one gcode line where theta is getting the wrong sign because the endingAngle is less than the startAngle even though our direction is positive. See line#45 in the file I posted [above.](https://github.com/MaslowCNC/Firmware/issues/426#issuecomment-379847129)
 Changing the sign of theta doesn't help.  I can set theta to zero, which results in a zero-length cut segment and a several-mm position error when the next commmand starts from the last known position on the line. This would be a good place to raise some kind of flag. Should I look for one or add one in the Report subsystem? Should the flag cause a 'Hold'?

BarbourSmith, 2018-04-10:
Good digging!

I'm still not sure I understand fully, is it a valid gcode that we are executing incorrectly, or is there an issue with the gcode itself?

Could we catch it on the Ground Control side and print a warning when the file is loaded to prevent someone from having to try to cut it?

blurfl, 2018-04-10:
The gcode is valid, and the parameters would be valid (I think) if we weren't running into the limits of floating point representation and math. It is when the values are approximated that the trouble arises.

The problematic line looks much like the previous two lines, both of which compute correctly. I don't know how you could detect the troublesome line without having GC 'do the math' before sending the line. I don't think GC has enough of the data to do that anyway.

BarbourSmith, 2018-04-10:
Hmmmm that is unfortunate. 

In that case would you recommend displaying a message when the issue happens so at least the user knows where the issue comes from?

blurfl, 2018-04-11:
Yes, I'm thinking that there should be a function somewhere to put up a message to flag bad gcode lines - not unimplemented ones, but ones like this that we can identify as causing a cut problem. Something like:

>"Message: Gcode error in this line - make a note of the line number. Cutting will be put on hold until you choose whether to 'Resume Cut' (skipping this bad line) or 'Stop'.       Maslow paused

I can see how to trigger a 'Hold' state after that dialog has been closed, but I can't get the 'Hold/Resume' button to switch back to ' II ' from 'Resume Cut' if the user chooses to live dangerously. Also wondering where a warnBadGcodeLine() belongs? Should it be a 'Message:' or an 'Alarm:'?

BarbourSmith, 2018-04-11:
That sounds perfect.

I vote for Alarm, seems more like an alarm to me.

It looks to me like the hold button should change it's color when data.uploadFlag changes, do you not find that to be the case?

```
def onUploadFlagChange(self, callback, newFlagValue):
        if self.data.uploadFlag is 0 and self.data.gcodeIndex > 1: #if the machine is stopped partway through a file
            self.holdBtn.secretText = "CONTINUE"
            self.holdBtn.btnBackground              = self.data.iconPath + 'ContinueYellow.png'
        else:
            self.holdBtn.secretText = "HOLD"
            self.holdBtn.btnBackground              = self.data.iconPath + 'HoldYellow.png'
```

blurfl, 2018-04-13:
In order to make this come up as an Alarm, the reportAlarmMessage() function in Report.cpp needs to be fleshed out, and GC's main.py needs to be made aware of serial traffic that begins "ALARM:". 
See Firmware PR#428  and GC PR#698

madgrizzle, 2018-04-16:
Does this alarm come mid cut?  Is it possible to do this when the file is loaded so someone starts to do the cutting?

Also, if the problem is with large arcs, is it possible to catch this situation and convert the gcode into a straight line in GC?

blurfl, 2018-04-16:
>Does this alarm come mid cut? Is it possible to do this when the file is loaded so someone starts to do the cutting?

It does come mid cut, during the calculations of an arc. I think it would need to be checked by the Arduino rather than GC. While it does show up if the Arduino is run in FAKE_SERVO mode, that takes essentially the same amount of time as running the cut does. A FAKE_SERVO setting to pre-flight a file would be possible but I wonder if it would be much used.

>is it possible to catch this situation

The problem isn't something I can tell how to identify in advance. Pick the problem line:

```
G3 X1.3116 Y8.3733 I1628.8304 J-1.1611
G3 X1.3115 Y7.1195 I9563.8567 J-1.9917
G3 X1.3114 Y6.6195 I9563.8569 J-0.7379
G3 X1.3114 Y6.3817 I9563.8569 J-0.2379
G3 X1.3116 Y4.39 I9563.8569 J0
```

It's the fourth one, all the others cut just fine.

> If you truncate the I and J values, does it convert it to a straight line? 

I tried various truncations with no joy. I also wondered about the advisability of changing a gcode without asking the user.

madgrizzle, 2018-04-16:
I can't say why the fourth line fails.  I would suspect the fifth line would be the most likely candidate on the basis that J is 0 and that if its involved with the denominator of a calculation, results can get crazy.

As for as truncations, are you saying that changing the fourth line from 
`G3 X1.3114 Y6.3817 I9563.8569 J-0.2379
to
G3 X1.3114 Y6.3817
`
still results in a fail?  Are you sure its the fourth line and not the fifth line>

blurfl, 2018-04-16:
There's no question that it's the fourth line. Setting I and J to zero creates a situation where the position error suddenly jumps to several mm. This trips the position error limit in PR#429...

BarbourSmith, 2018-04-16:
G3 X1.3114 Y6.3817 I9563.8569 J-0.2379 to **G1** X1.3114 Y6.3817

You would have to change it to a G1 command to make it move in a straight line, a G3 will always be an arc, if I or J are not specified they are assumed to be 0

blurfl, 2018-04-16:
> setting I and J to zero creates a zero radius curve which I guess results in the motors not moving. The motors don't move, but the PID loop expects the position to change, thus the position error.

>You would have to change it to a G1 command to make it move in a straight line

A great idea! That works pretty well. How to note the change in the log? Do we need to put up a Message: panel to notify the user?

BarbourSmith, 2018-04-16:
Under what circumstances would we make the change? I'm worried about changing someone's gcode...although I guess if the arc is so large that it is basically the same as a straight line it would be OK...what would the threshold be?

If we can find a good enough threshold I say we print a warning, but it doesn't need to spawn a popup in GC

madgrizzle, 2018-04-16:
I would suggest a setting that would enable automatic conversion and then print in the log where changes happen.  If disabled, then GC would prompt to have the user choose "continue", "convert once", "stop", or "all" ("all" being convert all for this file).

blurfl, 2018-04-16:
I've found that the problem [here](https://forums.maslowcnc.com/t/random-movements-while-cutting/3190/19) is again the precision thing, but I can identify it by looking at the height of the chord described by the G2/G3 command. Problematic commands have a chord that resolves to 0.01mm or less. I've tried trapping that situation and calling G1 instead with success. This turns an arc with less than 0.01mm of bow into a straight line.
 Making this change G2/G3 into G1 also fixes the file for this issue as well.
 I haven't tested it on any longer arcs, those in these files are a few mm long. I'm not set up to generate a test set of longer arcs. Does anyone have something I could test against?

BarbourSmith, 2018-04-16:
Here is a 200 inch circle with a 4x8 sheet added for scale:

[image]

[testarc.zip](https://github.com/MaslowCNC/Firmware/files/1917537/testarc.zip)

I wouldn't try to run it in real life 😉 but in the simulator it should be a good test to test against, but not quite what you are looking for. It seems like MakerCAM is just splitting it up into a lot of smaller arcs...which is normally a good thing...hmm

BarbourSmith, 2018-04-16:
Here's another giant arc:

[image]

This one is just one gcode line:

[testarc2.zip](https://github.com/MaslowCNC/Firmware/files/1917559/testarc2.zip)

But again as I'm posting this I don't think it's really what we need to test. In this case this is just a silly thing to test because one giant arc will be truncated to a single move directly to the right and since neither of those are valid movements in our machine they are both invalid.

What we really need is a file which IS reasonable to run, but which is made up of small segments of VERY large arcs...which is tough to create because it only really happens when the CAM program is behaving badly.

Maybe the original file which created this issue is the only good test we have?

blurfl, 2018-04-16:
That first circle is a pretty good test, MakerCam divided the circle into lots of little arcs, and those arcs are pretty flat. Here's an expurgated log of cutting up through the first ten arcs. 

[testarc-log.zip](https://github.com/MaslowCNC/Firmware/files/1917640/testarc-log.zip)

Ignore all the test-printing, the gist is that each G3 line involved a chord height of less than 0.01mm and was replaced with a G1 line.
The second one-line file cut the G3 as an arc because the chord height of that arc is 0.25mm.

What should be printed to the log?
> Oops! This gcode line has caused a calculation error, it will be replaced by: G1 X124.059 Y11.319 

Is too wordy and too informal.

BarbourSmith, 2018-04-16:
>  chord height

Cool vocab!!!

I think the error message should tell people a little bit more about what exactly is triggering the error. What are the criteria to see the error? Radius > X?

blurfl, 2018-04-16:
>What are the criteria to see the error?

[image]
It is the chord height of the arc _h_, calculated from the XY of the start and end points and the radius. That's how close  the arc is to being a straight line between the end points.

BarbourSmith, 2018-04-16:
Elegantly done! Also probably more than we want to explain in the warning message. Since we can't easily explain exactly what's being calculated I think:

> Oops! This gcode line has caused a calculation error, it will be replaced by: G1 X124.059 Y11.319

Is an excellent message to display

blurfl, 2018-04-17:
Were you eying the STATUS_GCODE_ARC_RADIUS_ERROR status message?
This will no longer bring up an ALARM: or Notification: panel, is that OK? We still have ALARM_GCODE_PARAM_ERROR which won't apply. Should I strip that back out of Report.h or leave it?
 If I create a STATUS_GCODE_CHORD_HEIGHT_ERROR and use Report.cpp::reportStatusMessage() I don't think I can put the G1 line into the message... Is having the G1 line important?

BarbourSmith, 2018-04-17:
I think it's OK that we don't bring up an alarm. In most cases people probably will never see anything, it will just work right. If something does go wrong we'll have it in the log to investigate. 

I'd say leave the status messages in if you think you will use them in the future, but if we don't need them anymore no reason to keep them, right?

blurfl, 2018-04-17:
Doing further testing I found that a file with 'almost complete' circles fell through the crack with the chord height test, so I added a test of the angle. The test is now "if chord height is less than 0.01mm and the arc is less than 0.5 radians". That patch makes nearly full circles cut properly.

<b>There is still something wrong</b>, though, with a sequence like this:
```
G1 X-2 Y0 F20
G3 X-1.85 Y1 I-8.0 J1.0
G1 X-2 Y0
G3 X-1.85 Y1 I-8.0 J-1.0
```
the first arc renders and cuts correctly, the second one renders wrong and posts a position error.
Here is GC's rendering
[image]

and here is g-code simulator
[image]
The firmware sends the bit toward the endpoint shown in the g-code simulator before raising the position error.
[image]

This behavior is present with the v1.11 release firmware as well, though it lacks the position alarm, so the bit jumps from the position shown immediately above to the position GC (incorrectly)  indicates for the endpoint. Puzzling indeed.

madgrizzle, 2018-04-17:
I'm confused about the gcode, must not understand it properly, but I thought the X and Y values were the end points.  If my understanding is correct, shouldn't both arcs start at -2,0 and end at -1.85,1..

blurfl, 2018-04-17:
>If my understanding is correct, shouldn't both arcs start at -2,0 and end at X-1.85Y1

There's a good summary [here](https://www.tormach.com/g02_g03.html) in the section 'Center Format Arc:'. I haven't dug in yet, but I wonder whether we're interpreting I and J in'Incremental IJ mode' or 'Absolute IJ mode'. I'll open a new issue on this, not sure whether GC of Firmware.

madgrizzle, 2018-04-17:
Well, I read that and now my head is spinning.. but still it seems that the the X&Y values should still be the end point.  I wonder if there's a bug in the gcode viewer because what you show in GC seems to make sense to me.

blurfl, 2018-04-17:
It looks to me like 'Incremental IJ mode' is what you're describing and kind of what GC is mis-rendering and the firmware is expecting, but 'Absolute IJ mode' is what g-code simulator is rendering and the path that the bit is traveling. 
A basket of snakes.  It _had_ to be snakes...

madgrizzle, 2018-04-17:
That g-code simulator image you posted wasn't of the code you posted, was it (it had another pair of arcs in it)?  When I put the code into g-code simulator, I was getting confused by the magenta line (which looked like your left hand arc) but that was just the router bit or whatever.  Sorry for the digression.

blurfl, 2018-04-17:
The code in the pictures is this, which has a shorter chord height
```G20 G90
G1 X-2 Y0 F25
G03 X-1.95 Y1.5 I-8.0 J01
G1 X-2 Y0 F25
G03 X-1.95 Y1.5 I-8.0 J-01
G4 S10
G1 X-2 Y0 F25
M2
```

blurfl, 2018-04-17:
The answer is that the line with the negative J value is in relative mode. g-code simulator is smart enough to identify that and move the endpoint relative to the center. Let's move the discussino to the new isue opened [here](https://github.com/MaslowCNC/Firmware/issues/431#issuecomment-382120943).

blurfl, 2018-05-11:
I think we can close this, yes?

BarbourSmith, 2018-05-11:
I believe that we can 😀 

Good work!! 👍 👍

closed 2018-05-11 by blurfl