MaslowCNC/Firmware #435: Error at compiling various firmware versions

OliverGonzalezLeon, 2018-04-21:
Hello,
Just received my kit and wanted to transfer firmware to arduino but there are libraries missing.
 In V1.11 the error is:

_> Arduino:1.6.2 (Mac OS X), Placa:"Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"
> 
> In file included from TimerOne.cpp:39:0:
> Maslow.h:27:19: fatal error: Servo.h: No such file or directory
>  #include <Servo.h>
>                    ^
> compilation terminated.
> Error de compilación_

What to do? thanks

blurfl, 2018-04-21:
All the libraries should be in the cnc_ctrl_v1 folder along with the .iso. Leave everything in that folder. If you’ve moved anything from there, either put it back or unzip it again for a fresh start. 
 See if that helps.

OliverGonzalezLeon, 2018-04-21:
Hello Scott,

The problem was that I had an older version of Arduino IDE. Once I
downloaded the last version, everything is running.

I even connected the motors and did the testing from the PC sofware and are
running!!

Thanks,

Oliver

blurfl, 2018-04-21:
That’s great to hear!👍😄 Thanks for coming back with the solution. If it’s ok with you, I’ll close this issue?

closed 2018-04-22 by blurfl

esspe2, 2019-04-28:
See also issue #491:
Servo.h: file not found on arduino 2.*