MaslowCNC/Firmware #436: Add support for configuring spindle control signal high/low.

CptanPanic, 2018-04-23:
Currently I am setting up my spindle control logic to connect to a super-pid spindle controller.  It uses an active low to trigger it, and I notice in code that there is almost support to have option to have active low or high but it is hardcoded to be `boolean activeHigh = true;`   Enhancement would be to add ability to pull this value from sysSettings  , and a '$' command to set it.

Thanks.

CptanPanic, 2018-04-23:
I am going to make this and submit a PR.

blurfl, 2018-04-23:
Instead of adding another setting, how about changing the existing setting for 'Spindle Automation' to a pick list with ['None', ''Relay - Active High', 'Relay - Active Low', 'Servo Operated Switch'] and implementing the logic in Spindle.cpp for all those options. That way non-spindlers won't have to wait for a function they don't have...

CptanPanic, 2018-04-23:
Yeah that sounds good

blurfl, 2018-04-23:
Make 'none' the default.
Don't configure the gpio as an output until an M-code that actually turns the spindle on is executed. This will aid to prevent turning on the spindle on Arduino reset. Any other precautions to prevent inadvertent activation would be a very good idea. Make it safe.

closed 2018-04-24 by BarbourSmith

blurfl, 2018-04-25:
```
what happens when the arduno looses power or reboots, if that's
going to cause the router to start spinning
```
Unfortunately, I don't think we can't know that. There are many ways to use a gpio pin to cause a change in an external circuit. Few people consider failure modes when lashing up a control like this.

CptanPanic, 2018-04-25:
Since pin isn't written to until M3 is sent, it seems that the internal resistor is activated by default on boot so the active low isn't triggered, at least for my device.

PR #439 (2018-04-24): Added support for active low spindle automation.
I change the spindleAutomate settings to be an enum, so now you can be values below.  Note this will require a change in ground control, as it setup as a bool currently.  

```
enum SpindleAutomationType {
  NONE,
  SERVO,
  RELAY_ACTIVE_HIGH,
  RELAY_ACTIVE_LOW };
```

closes #436