MaslowCNC/Firmware #459: Error Compiling

ebichan21, 2018-08-14:
I downloaded and redownloaded the latest IDE controller and firmware today. When I try to upload the firmware to the board, I get the following message:

Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

c:\users\ebich\appdata\local\arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2\bin\../lib/gcc/avr/4.9.2/../../../../avr/bin/ar.exe: unable to rename 'core\core.a'; reason: File exists

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

I don't know if this means it actually worked the first time, but I got the same message then.

I don't think the upload of the firmware worked since when I tried and use Ground Control, which I also downloaded today, McAfee quarantines it after ground control tells me there is no device.

As I have never used Arduino before, I have no idea what I'm doing.

ebichan21, 2018-08-14:
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\Settings.cpp:24:0:

C:\Users\ebich\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\EEPROM\src/EEPROM.h:43:30: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]

     operator const uint8_t() const       { return **this; }

                              ^

C:\Users\ebich\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\EEPROM\src/EEPROM.h:92:26: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]

     operator const int() const          { return index; }

                          ^

sketch\System.cpp:254:43: warning: unused parameter 'aux3' [-Wunused-parameter]

 void configAuxLow(int aux1, int aux2, int aux3, int aux4, int aux5, int aux6) {

                                           ^

sketch\System.cpp:254:63: warning: unused parameter 'aux5' [-Wunused-parameter]

 void configAuxLow(int aux1, int aux2, int aux3, int aux4, int aux5, int aux6) {

                                                               ^

sketch\System.cpp:254:73: warning: unused parameter 'aux6' [-Wunused-parameter]

 void configAuxLow(int aux1, int aux2, int aux3, int aux4, int aux5, int aux6) {

                                                                         ^

sketch\System.cpp:262:24: warning: unused parameter 'aux7' [-Wunused-parameter]

 void configAuxHigh(int aux7, int aux8, int aux9) {

                        ^

sketch\System.cpp:262:34: warning: unused parameter 'aux8' [-Wunused-parameter]

 void configAuxHigh(int aux7, int aux8, int aux9) {

                                  ^

sketch\System.cpp:262:44: warning: unused parameter 'aux9' [-Wunused-parameter]

 void configAuxHigh(int aux7, int aux8, int aux9) {

                                            ^

c:\users\ebich\appdata\local\arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2\bin\../lib/gcc/avr/4.9.2/../../../../avr/bin/ar.exe: unable to rename 'core\core.a'; reason: File exists

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

BarbourSmith, 2018-08-14:
> Try creating a new minimal sketch, File/new and compile that. If that compiles, see if now the firmware will compile.

I think that is an excellent suggestion. You can even just click **File -> New** and try to compile and upload the blank sketch that it creates. 

I would also try uninstalling and reinstalling the Arduino IDE, we've seen that fix a couple weird bugs

sullit, 2018-08-15:
From that pathing output it look's like you're using the Windows10 app ... ?

Try downloading and installing the 1.8.5 IDE in the c:\arduino\ path and you should be good to go.

blurfl, 2018-08-16:
@ebichan21 , have you resolved this? Can this issue be closed, or is more help needed?

ebichan21, 2018-08-16:
Well, the minimal sketch worked like a charm... I think. The Arduino IDE says it loaded, BUT Ground Control is still popping up like a virus AND when I turned off the virus scanning temporarily, I still get an error message. I'll try to type it in here, but it disappears and the Ground Control shuts down before I can read all of it.

ebichan21, 2018-08-16:
The error I get is:
Unable to find valid machine position for chain lengths 0.00, 0.00. Please set the chains to a known length (actions >Set Chain Lengths)
It took about 20 or 30 times loading it as it flashes on and then the whole program shuts down. I get this message as soon as I select the port. 
As I'm going back through the instructions, it lists assembling the electronics, then programing the Arduino, then taking care of ground control, then building the frame, and finally the sled, but nowhere is there a time for the gears, chains, etc. I would assume this is during the frame build, which is after ground control, but am I wrong? Also, what parts go where for the motors, chain, elastic cord, etc. I remember seeing this a while ago, but I don't see it now that I have time and space to set up the machine.

ebichan21, 2018-08-16:
I can tell where some go, like the gears on the motors, but I'm not sure for all the parts.

blurfl, 2018-08-17:
Good to hear! I'll close this issue then.

closed 2018-08-17 by blurfl