MaslowCNC/Firmware #487: error timeout when downloading audrino

oscar5208, 2019-03-18:
while downloading aurino to laptop got below error?can i download all software to my main pc then put sketch on thumb drive for laptop to use on cnc machine or does all 
[image]
software have to be on laptop attached to machine?
[image]

GeroBH, 2019-03-18:
Responded to the post in the Forum, as this does not seem to be an Firmware issue.
https://forums.maslowcnc.com/t/laptop-set-up-with-audrino-software-download/9186

If you feel OK to continue the troubleshooting in the Forum, please close this issue here on Git.
Regards, Gero

blurfl, 2019-05-24:
Can we close this?

BarbourSmith, 2019-05-24:
Yeah I think it is safe to close this

closed 2019-05-24 by BarbourSmith