MaslowCNC/Firmware #491: Servo.h: file not found on arduino 2.*

esspe2, 2019-04-28:
Hi,

I have built the firmware 1.26 successfully on a machine with Aduino IDE 1.8.x, but it failed with Arduino 2.x.
Apparently Servo.h is "included" in the new versions of arduino and doesn't need #include line, so a crude workaround is to comment out that line in Maslow.h.
A better change would be to add something like this:
#if (ARDUINO < 2)
#include <Servo.h>
#endif

Maybe other changes could be necessary, but the easiest way for now is to stick with the 1.8.x IDE.

Still need to check before doing a PR.

Regards, S. Peter's.

blurfl, 2019-04-28:
I've built successfully with the Arduino IDE v1.9.x branch, but never saw a 2.x branch. Is there a link to it?

esspe2, 2019-04-28:
Well pointed! 
the version number is exactly : 2:1.0.5+dfsg2-4 installed from the ubuntu repository with apt-get.
The IDE seems older, and files are dated 9 dec 2016.
So at the end maybe it was just a 1.0.5 version , still working but outdated.

And the solution is to uninstall this version and download the desired one from the Arduino website.

Regards, S. Peters.

closed 2019-04-28 by esspe2