MaslowCNC/Firmware #497: Arduino missing cncctrlv1.ino files

killswitchKSE, 2019-05-05:
Been unable to calibrate the maslow cnc due to the arduino download missing the cncctrlv1.ino files. I haven't seen a single place that offers help on the issue. I've download and installed the arduino, firmware, and ground control software several times but still cant get the system to work; When trying to calibrate the maslow the motors wont turn apparently due to the arduino not running the cncctrlv1.ino as the files are missing. I dont know if older versions of the arduino software has those missing files or if its even possible to download an older version. 

Help would be much appreciated and yes all files are the latest updated version also i'm running win 10 64bit pc

BarbourSmith, 2019-05-06:
Hmmm I haven't seen an issue like that before. cnc_ctrl_v1.ino seems to be in the proper place to me, you can see it here: https://github.com/MaslowCNC/Firmware/blob/master/cnc_ctrl_v1/cnc_ctrl_v1.ino the only thing I can think is that maybe your download has become corrupted and it might be worth downloading the folder again? Are you sure about the file name ` cncctrlv1.ino` that is without the `_` characters?

Can you post a copy of the error message you are getting or a screenshot?

killswitchKSE, 2019-05-07:
I appreciate the help... Problem was resolved as i was skipping over a step; probably because i wasn't feeling well the day when i was setting it up but today i quickly realized that i wasn't checking the firmware folder for the cncctrlv1.ino file... Thanks again for you help

BarbourSmith, 2019-05-08:
Excellent! That is great to hear 👍 👍

closed 2019-05-08 by BarbourSmith