MaslowCNC/Firmware #506: 'RingBuffer' conflicts with definition in ARDUINO_GRAND_CENTRAL_M4

blurfl, 2019-05-27:
Working on porting to the Grand Central M4, and I've run into a name collision between a definition between the definition of Ringbuffer in 
Arduino15/packages/adafruit/hardware/samd/1.4.1/cores/arduino/RingBuffer.h
and in the Maslow firmware. 

 I'm not sure of the best workaround. Globally changing the Maslow 'RingBuffer' to 'ringBuffer' is one workaround, but is there a different or preferred way?

BarbourSmith, 2019-05-27:
> Globally changing the Maslow 'RingBuffer' to 'ringBuffer' is one workaround, but is there a different or preferred way?

That sounds completely OK to me. We could even call it maslowRingBuffer if we really want to be sure of avoiding conflicts in the future.

blurfl, 2019-05-27:
I'll put in a PR to name it maslowRingBuffer, thanks for the quick response!

BarbourSmith, 2019-05-27:
😀 Sounds fantastic