MaslowCNC/Firmware #509: Open GL issue

TheSawBoss, 2019-06-06:
Hi, new to this.  Trying to setup the Ground Control, and I get an error that I have Open GL 1.1 and need open GL 2.0.  I actually have open GL 3.1 and just verified this fact with the OpenGL Extensions Viewer (see pictures).  I'm completely stuck on what to do.  Help!

[image]

[image]

BarbourSmith, 2019-06-06:
Hmmmm that is interesting that it is having trouble detecting the open gl version because you clearly have a later version.

How technical are you. Do you feel up for trying to run Ground Control from the source code by following the instructions here: https://github.com/MaslowCNC/GroundControl/wiki/Windows#how-to-install-and-run-ground-control-from-the-source-code ?

TheSawBoss, 2019-06-06:
Wow what a fast reply.  I'm not all that savvy but I'll give it a shot and let you know how it goes! thanks

TheSawBoss, 2019-06-08:
So I got to the last step and got stopped.  I'm hoping this is just from my own ignorance rather than something more complicated.   Any suggestions?  Also, I added "'C:\Python27\Scripts" to the PATH, as suggested by the error message.
[image]
[image]

BarbourSmith, 2019-06-09:
I think you are very very close! You need to use the CD (change directory) command to go to the directory where the main.py file for Ground Control is. If you haven't already done so you can download a copy of the ground control files using the green download button on the upper right of this page: https://github.com/MaslowCNC/GroundControl

TheSawBoss, 2019-06-09:
Sweet, that seemed to work, and now I am in Ground control, within the DOS screen.  It clearly recognized my OpenGL 3.1, but then when I launch Ground Control file I get the same error (note that I launched the Ground Control application from the original "GroundControl-Windows.Portable.v1.26" folder that was downloaded, as the one you referenced above "Ground Control Master" doesn't seem to have a launch file).
[image]

[image]