MaslowCNC/Firmware #517: An error occurred while uploading the sketch

Ozista, 2019-07-25:
Hi, I have just downloaded the latest (2 days old) new version of "cnc_ctrl_v1" Maslow firmware.  I have an AT Mega 2560 Chinese board. While uploading the sketch, the diodes flashing for 1 second, then long after (<15 sec) the message  "An error occurred while uploading the sketch".

I have successfully uploaded the Exaplesample/Basic/Blink sketch from the file menu of my Arduino Ver. 1.8 9 and my USB port used is #9.

Thank you in advance for any ideas on the above error.

blurfl, 2019-07-25:
I get that error message when I forget to quit the GroundControl program before programming the Arduino. Both programs try to use the serial connection to the Mega. See if that's the issue, let us know what you find.

Ozista, 2019-07-26:
Hi Scott,

Thank you for the advice, I have made sure that the GroundControl was not
running as I heard about before.   I was lucky once with the Leonardo
controller but the capacity of the card is not enough, the sample "Blink"
is downloading on "Mega 2560" before and after the "Maslow Control
Software" with no problem.
My original 3D printer sketch "Marlin" is uploading fine but just not the
"CNC_CTRL_v1".
[image: image.png]

Istvan Toth

blurfl, 2019-07-26:
OK, sorry that didn't help. Does the firmware compile correctly? Can you copy the contents of the compile window into a .zip file and attach it here?

davidelang, 2019-07-26:
are you sure that it compiled properly?

David Lang

Ozista, 2019-07-26:
Good idea, the firmware compiles great thanks! Attached 2 files -with and without the upload Verbose.

Thank again.

[MaslowVerbozeOutput Compile-only.txt](https://github.com/MaslowCNC/Firmware/files/3435863/MaslowVerbozeOutput.Compile-only.txt)
[MaslowVerbozeOutput Compile-Upload.txt](https://github.com/MaslowCNC/Firmware/files/3435864/MaslowVerbozeOutput.Compile-Upload.txt)

blurfl, 2019-07-26:
It looks like it compiles correctly, but both files seem to be the output of the compile command rather than the upload. The output with upload Verbose should have a section at the end showing the output of the avrdude uploader. This is what it looks like here:

[avrdude-out.txt](https://github.com/MaslowCNC/Firmware/files/3436185/avrdude-out.txt)

Ozista, 2019-07-28:
Sorry, the upload tail of the verbose did not come with the previous file.  Please find attached again and the screenshot of the settings while uploading.

Thank you for your time, Scott!

[image]
[MaslowVerbozeOutput.Compile-Upload.txt](https://github.com/MaslowCNC/Firmware/files/3438779/MaslowVerbozeOutput.Compile-Upload.txt)

blurfl, 2019-07-28:
I think the programmer should be "AVR ISP" - that's what shows up here. If that doesn't fix things, let's look at the verbose capture from uploading Blink.ino.

Ozista, 2019-07-28:
I think you're right there Scott, " the programmer should be "AVR ISP" "  I
used an AT Tiny board and lost the tread back to which programmer I had for
Mega 2560. Tried before, it seems to be no difference in uploading
MaslowCNC today.. but Blik is all good with  "AVR ISP" too.

Istvan Toth

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Isti\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=1A86_7523 -ide-version=10809 -build-path C:\Users\Isti\AppData\Local\Temp\arduino_build_746435 -warnings=none -build-cache C:\Users\Isti\AppData\Local\Temp\arduino_cache_462503 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -verbose C:\Users\Isti\Downloads\Firmware-master (2)\Firmware-master\cnc_ctrl_v1\cnc_ctrl_v1.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Isti\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=1A86_7523 -ide-version=10809 -build-path C:\Users\Isti\AppData\Local\Temp\arduino_build_746435 -warnings=none -build-cache C:\Users\Isti\AppData\Local\Temp\arduino_cache_462503 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -verbose C:\Users\Isti\Downloads\Firmware-master (2)\Firmware-master\cnc_ctrl_v1\cnc_ctrl_v1.ino
Using board 'mega' from platform in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\Isti\\AppData\\Local\\Temp\\arduino_build_746435\\sketch\\cnc_ctrl_v1.ino.cpp" -o nul
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\\hardwa [... cut ...] re\\arduino\\avr\\variants\\mega" "C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\cores\\arduino\\HardwareSerial.cpp" -o "C:\\Users\\Isti\\AppData\\Local\\Temp\\arduino_build_746435\\core\\HardwareSerial.cpp.o"
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-stat

blurfl, 2019-07-28:
Sorry, I can't make much of the huge text dump in your last message. Could you edit that message to remove the text dump (to tidy up the thread) and attach the dump as a text attachment?
 I still don't see the output from the avrdude uploader. In Arduino preferences, turn off 'verbose output during compilation', turn on 'upload'...

Ozista, 2019-07-29:
Sorry, that is maybe due to replying within the email platform.
Hope these attachments will appear separately?!

[MaslowVerbozeOutput.Compile-Upload2.txt](https://github.com/MaslowCNC/Firmware/files/3439821/MaslowVerbozeOutput.Compile-Upload2.txt)
[Blink Copile-Upload verbose.txt](https://github.com/MaslowCNC/Firmware/files/3439822/Blink.Copile-Upload.verbose.txt)
[image]

blurfl, 2019-07-29:
Searching online for 'arduino avrdude: stk500v2_ReceiveMessage() timeout' brings up some things that might apply.
>Try to reset the AVR (press and release the button) while the programmer is attempting to connect.
>There has been problems with the version from the Microsoft Store. Try to remove the old version and install the newest: arduino.cc/en/Main/Software
>Having three '!' characters in a row in the source file  - !!! - can cause problems for the bootloader

Ozista, 2019-07-29:
Wow, Scott that was the -!!!- problem! Thank you so much! I could upload after finding "!!!" and replaced all to one "!". 
I do not know how many and/or where they were but the Ctrl F worked to find and the Replace all in all sketches at one...

Thank again Mate, hope I can help in some way soon.

Istvan Toth

blurfl, 2019-07-29:
🎉👍 That sure was a strange one! Glad it’s solved.

BarbourSmith, 2019-07-29:
Nice work!!!