MaslowCNC/Firmware #520: make B11 able to control both motors in one command

davidelang, 2019-08-05:
If I understand the login in reading the b11 code, you can't issue commands for both left and right motors at the same time. Please consider fixing this.

blurfl, 2019-08-05:
Is the desired operation that the two motors run simultaneously, or sequentially? If sequentially, should they run in the order listed in the gcode line?

davidelang, 2019-08-05:
simultaniously, if you say to run both motors at 50% for 3 seconds, both motors 
should move at the same time and stop 3 seconds later.

ideally the times could be different, but it would be a big step forward to 
allow two power settings and only one time setting.

(also, I think I made a typo and this is B10,not B11)

David Lang

blurfl, 2019-08-05:
Presently, B11 runs a motor in the given direction at the given speed for the given time. Modifying that to run both motors simultaneously looks simple, I've ginned it up and tested it.

Presently B10 reports the encoder value of the axis specified:
```
B10 L
[Measure: 1756.27]

B10 R
[Measure: 1757.54]
```

davidelang, 2019-08-06:
one of the things on my (very long) todo list is to make versions of the B commands that let you move and report by encoder counts, not distance. I really think the internal work should be happening by encoder counts as integer values, only converting to/from real-world distance when evaluating g-code or reporting