MaslowCNC/Firmware #523: Default sys.feedrate = 1 mm/min

madgrizzle, 2019-08-13:
In testing the ability to send custom gcode commands (i.e, not loading a gcode file.. manual commands), I sent the following to the controller upon startup...

G1 X10 Y10

... and the sled moved incredibly slowly.  Within the G1 command, the feed rate is extracted from the gcode line, but since it was missing, it used what was currently in the sys.feedrate variable,  which was 0 because this variable apparently never gets initialized with a default setting.  But then, a few lines later sys.feedrate is constrained to be between 1 and sysSettings.maxFeed and therefore gets set as 1 (mm/min).  This explains the incredibly slow movement.

Normally this is not an issue since well generated gcode files include a feedrate.  But my concern is that some people at some time *cough* might do some manual commands and it not dawn on them to put in a feed rate if they use the G1 command and they hadn't sent something earlier with an actual feed rate.  So the question is whether or not we need to set a default for sys.feedrate that is, in effect, higher than 1 mm/min.. which is ridiculously slow.

blurfl, 2019-08-13:
I’ve been bitten by this a number of times. Not sure what value would make a default, though. 50% of full speed, maybe?

madgrizzle, 2019-08-13:
That seems reasonable.. just something that makes the sled visibly move.

davidelang, 2019-08-13:
just as a point of reference, if you tell grbl to move and have not given it a 
feed rate since it booted, it will just return an error.

David Lang

madgrizzle, 2019-08-13:
The firmware "had" provisions for reporting an error (see report.cpp) but it was all commented out.

BarbourSmith, 2019-08-13:
I think setting the default to something around 50% like maybe 400 makes a lot of sense... although we need to be sure not to be bitten by a mm/inch conversion there

madgrizzle, 2019-08-13:
see #524