MaslowCNC/Firmware #531: Does FAKE_SERVO need an alert when activating.

blurfl, 2019-08-29:
It's been less than a week, and a user has gotten into "FAKE_SERVO" mode [without realizing it](https://forums.maslowcnc.com/t/managing-motors-with-dual-l298n/9896/24). 

 I though that requiring a meta key to turn the function on, and printing the state would be enough to avoid inadvertent entry and indicate the change, but I may be wrong. 

 One option would be to put up a message box on entering the mode, with a note about how to get back out. That's complicated by the fact that the firmware controls the text of the message but cannot know the key used to trigger the state. Too, having that message box appear isn't needed for the intended user of "FAKE_SERVO" mode, only for explorers.

 Another option would be to add a switch to the GC settings to enable the FS option. That would further protect against inadvertent activation through GC.

 Other options? Is a 'fix' necessary?

BarbourSmith, 2019-08-29:
Hmmm tricky. What about a warning message that pops up if GC is launched in fake server mode? That might be easier to impliment and give folks a heads up about what might be going on

blurfl, 2019-08-29:
How hard would it be f\or GC to know its just <s>launched</s> connected and capture the "FAKE_SERVO ON" text from there firmware? Wish GC had a <s>startup</s> connect macro sequence...

@madgrizzle [suggested having a switch to enable it](https://forums.maslowcnc.com/t/make-fake-servo-a-firmware-setting-please/10891/34), separate from the choice of character. That has been pretty easy to set up:
[image]

What do you think about that? A refinement might be that the 'FAKE_SERVO_OFF' key always works to shut it off, but the '...ON' key only works if the '...allowed' setting permits.

madgrizzle, 2019-08-29:
I think you could capture that message in a couple places in ground control.  I think either in serialThread or the logger.  Just do a string comparison on every string that arrives and then do what you want with it.  I might do something like that in webcontrol and put up an icon to alert the user its in fake_servo mode.

As for your refinement, if for some reason you are in fake_servo and then disallow that mode, ideally it would put the controller out of fake_servo (issue the B99 OFF command).  But then you have to monitor settings and seems like a lot of work.  I suggest, as you suggested, that you should be able to exit fake_servo mode even if fake_servo is disallowed, but only enter it if fake_servo is allowed.

blurfl, 2019-08-29:
Part of the issue comes from having the setting persist over restarts. If that was removed, GC wouldn't need to watch and alert.

blurfl, 2019-08-29:
@madgrizzle, is having the setting persist over restarts a requirement for your use? This would be deafer without that...

madgrizzle, 2019-08-29:
So... if you are referring to something in groundcontrol, then don't worry about me.. I don't use GC.  But if you are asking if the firmware should revert back to 'off' mode when it restarts, I think that would be problematic for me.  When I develop, I stop and start webcontrol many, many times and it would be rather annoying to have to put it into fake_servo mode every time.

madgrizzle, 2019-08-29:
I think if someone 'allows' fake_servo mode to be enabled, then any issue they have is their own doing.  So putting in that switch, in my opinion, is good enough.  Don't try make it so idiot-proof that it defeats the purpose of doing it in the first place.. if you follow.

blurfl, 2019-08-29:
I think I can tighten up the EEPROM test, using some value other than '1' to indicate FS mode. Any suggestions?

madgrizzle, 2019-08-29:
Not sure what you are referring to.. but if you want another number to use, 42 is the ultimate answer.