MaslowCNC/Firmware #533: GCode won't run

richt218, 2019-09-09:
We have been using this program for a while, about 3 months, so we are fairly new and by know means tech savy.  Just went to try and cut out a new piece and the sled runs in a straight line (right off the sheet if we don't hit stop).

We set up Ground Control as usual, got the sled in the right position, load the GCode, and set up the home.  When we hit play the sled will go to the starting spot, lower the bit and then run in a straight line until we manually stop it in Ground Control.  

We have tried uninstalling and reinstalling all software and firmware but still get the same result.  We have tried using several different GCodes (both new and ones that have been successfully cut out) but can not seem to figure out what is going wrong.

Any help would be greatly appreciated.

davidelang, 2019-09-09:
can you post the g-code you are trying?

David Lang

richt218, 2019-09-09:
I am pretty new to this.  I am not sure how to do that.  I tried to drop the file in but it said it does not support that file type.  We use Carbide Create to write the GCode, if that helps.

blurfl, 2019-09-10:
Github only allows a few file types as attachments. You can compress the file and attach the resulting .zip file. 

 Another approach would be to copy the first 20 or so lines and paste them into a comment as a code fragment: between lines starting with   ```  (the character that shares the key with ~ ) before and after.

```
This line is formatted as a code fragment
```

richt218, 2019-09-10:
[Hexagon Gcode.zip](https://github.com/MaslowCNC/Firmware/files/3597992/Hexagon.Gcode.zip)
This has the original file with the design and the GCode.

blurfl, 2019-09-11:
I don't see anything in the file that looks out of place. Which version of GC and Firmware are you using? What was involved in 'got the sled in the right position' and 'set up the home'? Which way did the sled travel when the file began to run?

The .nc file looks good, it describes a hexagonal object about 82mm across with a hexagonal pocket cut down into it and tabs on three of the outside edges. The shape will be centered at about 40mm above and 50mm to the right of 'Home' which would be in the middle of the workarea in the normal setup, or in the location where the router was located when the 'Define Home' button in GC was clicked.

The file begins by setting metric coordinate mode, setting absolute positioning mode, and then sending the router to X0.0 Y0.0 Z10.0 .
```gcode
G21
G90
G0X0.000Y0.000Z10.000

```
The motion will be in all three axes at the same time, the bit rising or plunging to 10mm above the surface as it traverses to the 'Home' location. If the Home location has been moved from 0,0, GC will translate the XY coordinates to the new Home location.

richt218, 2019-09-12:
When I set everything up in Ground Control, I position the sled to a home location on the sheet where we have room to cut out the project.  I open the GCode and then "define home".  When I hit play, the router moves to its starting position, lowers the bit down to make the first pass, and then starts moving but only in a straight line.  The first time it went straight left almost off the sheet.  The second time it went straight up until we stopped it. It has gone in every direction.  After reinstalling everything on my laptop and trying to do a cut, the router was set to cut on the left side of the sheet, but instead traveled to the center of the sheet and then went back to the left past where it was supposed to cut and almost off the sheet before we stopped it.

blurfl, 2019-09-12:
Does the sled respond as expected to the on-screen arrow controls, moving the distance shown next to the 'Dist To Move: ' box?

richt218, 2019-09-12:
Yes the sled follows the commands to move on-screen.

blurfl, 2019-09-12:
This is a good puzzle. When your sled is at the 'Home' position you've chosen for this cut, what are the X: and Y: coordinates on the screen? Can you post a screen shot?

[image]

davidelang, 2019-09-12:
can you stop groundcontrol, clear the log file, start groundcontrol and then try this file, then post the log file.

I'm suspecting that something in it is confusing the parser, hopefully it's logging an error.

this g-code includes things like motor control that the maslow doesn't support. what is being used to generate this?

richt218, 2019-09-15:
blurfl:  Here is a screen shot once home is defined.
[image]

richt218, 2019-09-15:
blurfl: How do I clear the log file?

blurfl, 2019-09-15:
Thanks for uploading the screen shot, I think I see whats going on here. The file 'Hexagon Gcode.nc' uses metric coordinates, but from the screen, GC looks like it's been set to use inches coordinates. GC is interpreting the gcode lines as inch measurements instead of millimeters. I've set up here in the same way and can repeat the issue you describe, so I'll look further into it. I think GC can get confused when the 'Define Home' routine is run in inches when the file is written for metric.

 In the mean time to get you cutting right now, once you've got your chains recalibrated stay in metric mode to move the sled and run the file. I think that will operate as expected, not heading off the sheet.

richt218, 2019-09-16:
We got everything set up.  The sled follows commands to move on screen.  We tried to run another cut but the same thing keeps happening.  The sled moves in a straight line in one direction and continues until we stop it.  We did notice in the log that the x axis jumps to a really large number to a spot way off the page.  We tried it separate times and got the same result each time.  I took screen shots so you can see what we see.
[image]
[image]
[image]

blurfl, 2019-09-17:
Ok, let's get the log file to see what that shows. Are you using Windows, MacOS or Linux? The log file is in a different spot for each...

richt218, 2019-09-19:
Windows

richt218, 2019-09-24:
Here are the log files that I could find.
[log 2.zip](https://github.com/MaslowCNC/Firmware/files/3649694/log.2.zip)

richt218, 2019-09-30:
Are these the correct logs or do I need to look in a different spot for them?