MaslowCNC/Firmware #536: error when trying to update the firmware on Arduino (Maslow cnc)

3dlines, 2019-11-08:
Hi all, 

I need your assistance with setting up the Maslow as the firmware won't update and it keeps giving me errors when trying to upload to Arduino. 
I copied the errors it gives me, thanks for any insight. 

"Arduino: 1.8.10 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

cnc_ctrl_v1:1:1: error: expected unqualified-id before numeric constant

 0/*This file is part of the Maslow Control Software.

 ^

In file included from C:\Program Files (x86)\Arduino\libraries\Servo\src/Servo.h:63:0,

                 from sketch\Maslow.h:27,

                 from C:\Users\benbo\Desktop\firmware maslow\Firmware-1.26\cnc_ctrl_v1\cnc_ctrl_v1.ino:36:

C:\Program Files (x86)\Arduino\libraries\Servo\src/avr/ServoTimers.h:39:68: error: 'timer16_Sequence_t' does not name a type

 typedef enum { _timer5, _timer1, _timer3, _timer4, _Nbr_16timers } timer16_Sequence_t;

                                                                    ^~~~~~~~~~~~~~~~~~

Multiple libraries were found for "Servo.h"
 Used: C:\Program
Multiple libraries were found for "EEPROM.h"
 Used: C:\Program
exit status 1
expected unqualified-id before numeric constant

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

blurfl, 2019-11-08:
> 0/*This file is part of the Maslow Control Software.
>^
It looks like there is a stray character ( ‘0’) At the beginning of the very first line in the file ‘cnc_ctrl_v1.ino. If so removing the zero should solve the issue. Give that a try and let us know if that works.

3dlines, 2019-11-08:
Blurfl, 

Thanks for the response, I removed it but now I get a different error: 
Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

cnc_ctrl_v1:2:5: error: 'The' does not name a type

     The Maslow Control Software is free software: you can redistribute it and/or modify

     ^

cnc_ctrl_v1:7:31: error: 'without' does not name a type

     but WITHOUT ANY WARRANTY; without even the implied warranty of

                               ^

In file included from C:\Program Files (x86)\Arduino\libraries\Servo\src/Servo.h:63:0,

                 from sketch\Maslow.h:27,

                 from C:\Users\benbo\Desktop\firmware maslow\Firmware-1.26\cnc_ctrl_v1\cnc_ctrl_v1.ino:36:

C:\Program Files (x86)\Arduino\libraries\Servo\src/avr/ServoTimers.h:39:68: error: 'timer16_Sequence_t' does not name a type

 typedef enum { _timer5, _timer1, _timer3, _timer4, _Nbr_16timers } timer16_Sequence_t;

                                                                    ^

exit status 1
'The' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

blurfl, 2019-11-08:
Those errors make it sound like there's a missing comment delimiter '/*' at the beginning of a file.

It might be best to start with a fresh copy of the firmware files, available [here](https://github.com/MaslowCNC/Firmware/releases/latest).

3dlines, 2019-11-08:
I figured it out, I got this idea from you so thank you! 
I erased the two rows of the credits/copy rights from the author and it solved it - I am assuming something there wasn't written well.

closed 2019-11-09 by blurfl