MaslowCNC/Firmware #538: afmotorshield changes

tecan, 2020-02-20:
new port of maslowcnc to arduino motor shield v1 feel free to use the changes
https://github.com/tecan/MaslowCNC-ALT

BarbourSmith, 2020-02-21:
That is a super neat idea! Is that this motor shield: https://www.adafruit.com/product/81 ?

tecan, 2020-02-21:
yes same board, i hooked up the motors only today but im still getting everything calibrated and moving more pictures soon.

tecan, 2020-02-21:
the motors are moving but the encoders seem to need work still.

BarbourSmith, 2020-02-21:
That is fantastic.  If you have any encoder questions you will get a quick answer to them in the forums

davidelang, 2020-02-21:
On Thu, 20 Feb 2020, BarbourSmith wrote:

> That is a super neat idea! Is that this motor shield: https://www.adafruit.com/product/81 ?

that version has been discountinued (and would only support 0.6A per motor) the 
replacement is https://www.adafruit.com/product/1438 which has a PWM chip on 
board. It supports 1.2A per motor (up to 3A, but only for 20ms) so with the 
standard maslow hardware, it would go into thermal throttle relativly easily. 
but if someone had a version with less force required and keep the speeds down, 
they may get away with it.

this is usable on a 3.3v controller, which is good.

David Lang

tecan, 2020-02-23:
the motor shields are still available on ebay for 3 or 4 dollars. what i've seen sofar from the driver that it is even on USB 5v it was able to turn the motors well.

tecan, 2020-02-23:
motor 2 is spinning now too(pwm next).

tecan, 2020-02-23:
encoders are working but not keeping up..

20:14:41.463 -> ALARM: The sled is not keeping up with its expected position and has halted. Click the 'St0p' button to clear the alarm. More information at: https:/0github.com/MaslowCNC/Firmware/wiki/Keeping-Up  
20:14:41.497 -> [Forward 0alculating Position]
20:14:41.530 -> 0.00-54.000.00-54.000.00-55.000.00-56.000.00-58.000.00-59.000.00-57.000.00-55.000.00-53.000.00-51.000.00-49.000.00-47.000.00-45.000.00-43.000.00-44.000.00-46.000.00-48.000.00-49.000.00-51.000.00-52.00<Stop,MPos:0.00,0.00,0.00,WPos:0.000,0.000,0.000>
20:14:41.695 -> [PE:0.00,0.00,127]
20:14:41.729 -> 0.00-53.000.00-54.000.00-55.000.00-56.000.00-56.000.00-56.000.00-56.000.00-56.000.00-56.000.00-56.000.00-56.000.00-56.000.00-56.000.00-56.000.00-56.000.00-56.000.00-56.000.00-56.000.00-56.000.00-56.000.00,WPos:0.00[PE:0.0.00-56.000.00-56.000.00-56.000.00-56.000.00-56.006.000.000-56.00.6.000.00.00-56.0000.000>
20:14:42.126 -> [P0.00-56.000.00-56.000.00-56.000.00-56.00-0000-60.00<0,0.00,0.0[PE:0.00,-0.06,127]
20:14:42.358 -> 0.00-61.000.00-59.000.00-60.000.00-61.000.00-59.000.-59.3.000.00-51.000.00-49.000.00-50.000.00-56.000.00-58.000.00-60.00<Stop,MPos:0.00,0.00,0.00,WPos:0.000,0.000,0.000>
20:14:42.557 -> [PE:0.00,-0.06,127]
20:14:42.557 -> 0.00-61.000.00-59.000.00-57.000.00-56.000.00-58.000.00-60.000.00-62.000.00-64.000.00-66.000.00-68.000.00-67.000.00-65.000.00-64.000.00-66.000.00-65.000.00-64.000.00-66.000.00-65.000.00-64.000.00-63.000.00-62.00<Stop,MPos:0.00,0.00,0.00,WPos:0.000,0.000,0.000>
20:14:42.756 -> [PE:0.00,-0.08,127]
20:14:42.756 -> 0.00-61.000.00-59.000.00-58.000.00-60.000.00-62.000.00-64.000.00-66.000.00-67.000.00-65.000.00-66.000.00-67.000.00-68.000.00-70.000.00-71.000.00-69.000.00-67.000.00-65.000.00-66.000.00-68.000.00-70.000.00-72.00<Stop,MPos:0.00,0.00,0.00,WPos:0.000,0.000,0.000>
20:14:42.956 -> [PE:0.00,-0.16,127]
20:14:42.989 -> 0.00.0000.00-69.000.00-68.00.00-.0000.07.0000000-51.00<00,0.00[PE:0.00,0.02,127]
20:14:43.188 -> 0.00-49.000.00-48.000.00-47.000.00-45.000.00-43000050.0051.000.00,0.00,WPos:0.000,0.00[PE:0.00,-0.020.00-55.000.00-53.000.00-54.0055.000.00-53.000.00-54.000.000-4000,0.0[PE:0.00,-0.01,127]

BarbourSmith, 2020-02-24:
Nice work! Calibration is where I would start! 

You may find that with different motors you can move faster or slower than the default ones. There is a setting in Ground Control which will let you set the speed limit and make that error message go away if you still see it after calibration.

davidelang, 2020-02-24:
note that the "can't keep up" isn't just a motor thing, it takes time for the 
pid loops to ramp power up, and during this time the position lags behind where 
it should 'ideally' be, the default is a 2mm lag.

the real fix for this is acceleration planning so that it doesn't expect to 
start moving at full speed instantly and so it doesn't get upset when that 
doesn't happen :-)

In the meantime, at higher speeds, you will need to increase the error allowed 
in the settings.

David Lang