MaslowCNC/Firmware #542: New Feature: Z Axis limits to prevent equipment damage

ariasr, 2020-05-17:
I have made some small tweaks to the code to be able to set upper and lower limits for the z axis in order to prevent equipment strain.

I have added two variables to settings 
     float zAxisUpperLimit
     float zAxisLowerLimit

I have created 4 "B" commands
     B17 Set the z axis upper limit to the current z axis position
     B18 Set the z axis lower limit to the current z axis position
     B19 Clear z axis limits
     B20 Echo z axis limits

I have made some changes to Axis, GCode and Settings classes.

I would like to contribute it to the project.

Best Regards

davidelang, 2020-05-18:
once you get the robot e-mail, vote by adding a thumbs up to the robot's post, 
that will merge your pull request unless people vote thumbs down.

David Lang

On Sun, 17 May 2020, ariasr wrote:

> Date: Sun, 17 May 2020 16:10:39 -0700
> From: ariasr <notifications@github.com>
> Reply-To: MaslowCNC/Firmware
>     <reply@reply.github.com>
> To: MaslowCNC/Firmware <Firmware@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: [MaslowCNC/Firmware] New Feature: Z Axis limits to prevent equipment
>     damage (#542)
> 
> I have made some small tweaks to the code to be able to set upper and lower limits for the z axis in order to prevent equipment strain.
>
> I have added two variables to settings
>     float zAxisUpperLimit
>     float zAxisLowerLimit
>
> I have created 4 "B" commands
>     B17 Set the z axis upper limit to the current z axis position
>     B18 Set the z axis lower limit to the current z axis position
>     B19 Clear z axis limits
>     B20 Echo z axis limits
>
> I have made some changes to Axis, GCode and Settings classes.
>
> I would like to contribute it to the project.
>
> Best Regards
>
>
>
>

ariasr, 2020-05-18:
I have not been able to do a pull request. Perhaps I don't have the right permisions?

closed 2020-05-18 by ariasr

davidelang, 2020-05-18:
you should be able to fork the repo and create a pull request from your fork.

David Lang

On Mon, 18 May 2020, ariasr wrote:

> I have not been able to do a pull request. Perhaps I don't have the right permisions?
>
>

ariasr, 2020-05-18:
Thank you. I finally figured it out and generated the pull request. Its already there for aproval.

davidelang, 2020-05-18:
be sure to give yoursef a thumbs up (on the robot's post)

David Lang