MaslowCNC/Firmware #546: New release versions - stalled on 1.26 for over a year

gb0101010101, 2020-05-24:
Is there any reason why new versions of firmware are not being released?

Many Pull Requests (~40) have being accepted since v1.26 was released over a year ago. I have been compiling and using latest GitHub firmware code without any issues. The past few PRs had added features that require GroundControl/WebControl to check firmware version to see if the feature has been implemented. This cannot be easily done since the version number is not changing.

Please let me know if I can help in getting new releases out?

BarbourSmith, 2020-05-24:
Has it really been that many? I only remember one or two. I think the hold up is because a Ground Control release has to happen at the same time and I think the build process is broken there. I'll see if I can get that fixed on Tuesday and then we can try a release

gb0101010101, 2020-05-25:
> Has it really been that many?

Sorry quick count was off. Here are PRs merged since v1.26 (Jan 23, 2019)
https://github.com/MaslowCNC/Firmware/pulls?q=merged%3A%3E2019-01-24

It's crazy how time flies - especially this year!

> I think the hold up is because a Ground Control release has to happen at the same time

Perhaps we can unbind this requirement and add firmware version checking to GroundControl? Looks like most of WebControl is a copy of GroundControl so I could possibly backport code?

> I think the build process is broken there. I'll see if I can get that fixed on Tuesday and then we can try a release.

Great thanks.

BarbourSmith, 2020-05-27:
I'm having trouble getting the Ground Control build environment up and running again. I'll keep working on it, but I just wanted to update that I haven't forgotten

DavidRizzo, 2020-09-18:
Any updates on this?

BarbourSmith, 2020-09-19:
Unfortunately only bad news. I was unable to get the build process going or contact the person who used to run the OS X side of the build so I think we're stuck. Web Control has taken over as the primary interface software and the holey calibration firmware seems to be the best option these days.

Ground Control has been marked as outdated, should we do the same thing here?

davidelang, 2020-09-19:
the fact that webcontol has mostly replaced groundcontrol is a good reason to 
break the tie between the two. releasing the two in step was a good idea when we 
started it, but now that GC is unable to build, we should not hold up firmware 
fixes.

If we are to the point of saying that holey calibration should be the default, I 
think that would be a good thing, but I think there are other fixes that should 
be made available, so I would not rip out the prior calibration.

David Lang

  On Sat, 19 Sep 2020, BarbourSmith wrote:

> Date: Sat, 19 Sep 2020 13:19:45 -0700
> From: BarbourSmith <notifications@github.com>
> Reply-To: MaslowCNC/Firmware
>     <reply@reply.github.com>
> To: MaslowCNC/Firmware <Firmware@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: Re: [MaslowCNC/Firmware] New release versions - stalled on 1.26 for
>     over a year (#546)
> 
> Unfortunately only bad news. I was unable to get the build process going or contact the person who used to run the OS X side of the build so I think we're stuck. Web Control has taken over as the primary interface software and the holey calibration firmware seems to be the best option these days.
>
> Ground Control has been marked as outdated, should we do the same thing here?
>
>

BarbourSmith, 2020-09-19:
The issue with that is that anyone using Ground Control will see a warning when they try to use the latest firmware. That might actually the correct behavior, but I wanted to at least bring it up.

rspeed, 2022-01-04:
Now that GroundControl has been supplanted by Makerverse, what is the status of this repo? There's a fork at WebControlCNC/Firmware, but it isn't if/why that exists.

BarbourSmith, 2022-01-05:
Yeah it's the canonical origin. It also exists just to have a backup in case one of the other repos goes away or something.