MaslowCNC/Firmware #547: firmware error

cstanl3y, 2020-06-11:
After loading firmware selecting the board and com port and clicking upload tab I get the following error:
Arduino: 1.8.12 (Windows 10), Board: “Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)”

cnc_ctrl_v1:36:10: fatal error: Maslow.h: No such file or directory

#include “Maslow.h”

      ^~~~~~~~~~
compilation terminated.

exit status 1
Maslow.h: No such file or directory

This report would have more information with
“Show verbose output during compilation”
option enabled in File -> Preferences.

I went back and reloaded everthing again same error.

GeroBH, 2020-06-12:
Hi, this error shows up if the Firmware is not "un-ziped" / extracted before uploading to the Arduino. You should be able to solve it by extracting the compressed file to a folder of your choice and upload it from there.
[image]

cstanl3y, 2020-06-12:
Thank you. It’s been a learning curve for sure.

[screenshot]<https://user-images.githubusercontent.com/25812652/84467066-9cd32d00-ac83-11ea-85af-062f48c80c1c.jpg>

BarbourSmith, 2020-06-12:
Thanks @GeroBH ❤️ ❤️

closed 2020-06-12 by BarbourSmith