MaslowCNC/Firmware #553: Cannot get the Firmware to Upload to the M2 board.

MarkCorkins, 2021-10-09:
First, thanks for any and all help. Greatly appreciated.

Just received the M2 and I'm trying to upload the Firmware. I have several Raspberry Pi's and the instructions were pretty straight forth in this process.

When I selected the options in the guide and proceed, this is what I get. I tried several times to load this and I used two different computers. Both computers are Windows, one is ver10 and the other is ver7. Com3 was my port also.

Sketch uses 57294 bytes (22%) of program storage space. Maximum is 253952 bytes.
Global variables use 2009 bytes (24%) of dynamic memory, leaving 6183 bytes for local variables. Maximum is 8192 bytes.
Error downloading https://downloads.arduino.cc/packages/package_index.json

Arduino: 1.8.16 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Sketch uses 57294 bytes (22%) of program storage space. Maximum is 253952 bytes.

Global variables use 2009 bytes (24%) of dynamic memory, leaving 6183 bytes for local variables. Maximum is 8192 bytes.

Error downloading https://downloads.arduino.cc/packages/package_index.json

An error occurred while uploading the sketch

avrdude: stk500v2_ReceiveMessage(): timeout

avrdude: stk500v2_ReceiveMessage(): timeout

avrdude: stk500v2_ReceiveMessage(): timeout

avrdude: stk500v2_ReceiveMessage(): timeout

avrdude: stk500v2_ReceiveMessage(): timeout

avrdude: stk500v2_ReceiveMessage(): timeout

avrdude: stk500v2_getsync(): timeout communicating with programmer

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
----------------------------------------------------------------------

Any suggestions?

Thanks,
Mark

MarkCorkins, 2021-10-09:
Closed and fixed. Thanks BarbourSmith

BarbourSmith, 2021-10-09:
Happy to help 😃