MaslowCNC/Firmware #555: REQ: port to Due

TheOncomingStorm, 2022-12-06:
Until recently I was forced to deal with the not-remotely-great MakerMade Due controller that only accepts GRBL and doesn't work with GroundControl/WebControl. I've since gotten an original Maslow controller so I can calibrate much easier but I'm wondering how hard would it be to port the current code here to a Due? I tried loading it up in Arduino IDE and it complained about missing header files but that's as far as I got with it(admittedly, I've only been working on the idea for about 2 hours now). The way I look at it, if we can make the Due act like the Mega version then at least we could possibly get some additional speed and be able to calibrate better than MakerMade's idiotic version requires.

BarbourSmith, 2022-12-07:
I think it should be totally possible, and I'm happy to answer any questions that come up in the process. I can't think of any particular reason that it wouldn't work.

davidelang, 2022-12-07:
IIRC the biggest issue running on the Due is that it doesn't have any eeprom, that needs to be added (and the appropriate library used) so that it can keep track of where it is when it's powered down (so you don't have to do a calibration each time you power it up)

BarbourSmith, 2022-12-07:
Great memory @davidelang, I remember that being an issue now too. How does the Maker Made controller get around that?